emxsys/callattendant #13: Embellish Call Log web interface

emxsys, 2018-11-02:
- Highlight whitelisted and blacklisted callers
- Add capability to add caller to whitelist or blacklist

emxsys, 2020-07-26:
@zomGreg I'm going to try and wrap up the nav links and the basic "add to" blacklist/whitelist today.  Then release 0.2 will be complete.

zomGreg, 2020-07-26:
I've got nav links in https://github.com/zomGreg/callattendant/tree/frontend_testing if you want to see what I was doing. I'm no frontend wizard, so what you have will probably be better than that.

emxsys, 2020-07-26:
@zomGreg Excellent. I was going down the same path, but you're a step ahead of me.  I'd like take your changes in PR, please.

zomGreg, 2020-07-26:
okay, I'll pop them in.

emxsys, 2020-07-26:
@zomGreg The nav links work good. Thanks!  I'm embellishing the aesthetics and will check in the changes later today.

emxsys, 2020-07-27:
@zomGreg FYI: I found the changes made to `callattendant.css` are not realized until you clear your browser's cache/history.

emxsys, 2020-07-29:
The call log enhancements are complete for version 0.2.

closed 2020-07-29 by emxsys