emxsys/callattendant #116: +VSD Silence Detection command not working on Conextant-based modems

emxsys, 2020-11-04:
Problem: The modem, after answering a call, does not detect/report the silence after the caller hangs up. Upon investigation, I note the modem accepts, but does not apply, the provided silence detection settings.

Test details: I opened up a terminal session with the modem (using minicom on Linux) and attempted to set a silence detection of 5 seconds in voice mode. The modem accepts the `+VSD=128,50` command, but the silence detection remains disabled (0 seconds), as reported by the modem. This can be seen in the following minicom log, annotated with my comments.

---
Welcome to minicom 2.7.1

OPTIONS: I18n
Compiled on Aug 13 2017, 15:25:34.
Port /dev/ttyACM0, 13:33:04

Press CTRL-A Z for help on special keys

_I entered_ `atz` _to reset the modem_
`OK`

_I entered_ `ate1` _to echo my commands to the screen_
`OK`

`at`
`OK`

`at+vsd?` _# query silence detection settings before entering voice mode_
`ERROR` _# returned expected error: not in voice mode_

`at+fclass=8` _# enter voice mode_
`OK`

`at+vsd?` _# query silence detection settings_
`128,0` _# returned silence detection disabled: 0 secs_
`OK`

`at+vsd=?` _# query silence detection accepted parameter values_
`(0-255),(0-255)`
`OK`

`at+vsd=128,50` _# set silence detection to 5 seconds_
`OK`

`at+vsd` _# query silence detection settings_
`128,0` _**# silence detection is still disabled. Expected 128,50**_
`OK`

J0hnMatrix, 2020-11-05:
Hello,
Same result here:

```
Welcome to minicom 2.7.1

OPTIONS: I18n
Compiled on Aug 13 2017, 15:25:34.
Port /dev/ttyACM0, 14:56:17

Press CTRL-A Z for help on special keys

atz
OK
ate1
OK
at
OK
at+vsd?
ERROR
at+fclass=8
OK
at+vsd?
128,0
OK
at+vsd=?
(0-255),(0-255)

OK
at+vsd128,50
ERROR
at+vsd=128,50
OK
at+vsd?
128,0
OK
```
Event if you try another numbers, the result is the same:
```
at+vsd=127,70
OK
at+vsd?
128,0
OK
```

Maybe a kind of "commit" command needs to be type after the "at+vsd=xxx,xxx" or maybe this command is unsupported as the returned values are 0-255,0-255 for both sds and sdi?

emxsys, 2020-11-08:
@J0hnMatrix I'm testing the following block of code in `Modem.record_audio`.  Seems to work in bench tests. Will test and refine on a live phone system.

```python
 # Test for silence on Zoom modems (Conextant). The +VSD doesnt work on these modems.
 if self.model == "ZOOM":
     # Silent frames are made of a mix of \x80 and \x79 bytes, with 90% or more == \x80.
     count = audio_data.count(b'\x80')
     if count >= (CHUNK * 0.90):
         silent_frame_count += 1
     else:
         silent_frame_count = 0
     if silent_frame_count > 40: # 40 frames is ~5 secs
         print(">> Silent frames detected... Stop recording.")
         break
```

PR #127 (2020-11-10): Added silence detection for Conexant-based modems
Added the ability to detect silence for the Zoom 3095 modem and others that use the Conexant chipset.  
This feature allows recording to terminate after a short period after a caller hangs up.
- The new silence detection logic is used for all modems when recording audio
- The silence detection data provides the conditionals used to skip saving silent audio files.
- Side effect: also smoothed up the audio play back in play_audio 

Fixes #116
closed 2020-11-10 by emxsys