emxsys/callattendant #134: Make compatible with Ubuntu

emxsys, 2020-11-19:
Ensure the installation and runtime work on Ubuntu.  

- Disable interactions with the `gpiozero` library if the platform is not a Raspberry Pi.

advfr, 2022-09-10:
Hi Bruce ! First thank you for the amazing job you've done. Guess what ?! We also have robocalls in France. But also due to chip shortage it becomes difficult, if not impossible to get RPi. I tried to setup callattendant in a virtualenv on a Wyse thin client (amazing AMD cpu for less than 60$). Unfortunately it seems, as you posted initialy, that callattendant expect GPIOs to be there, which I have not, hence the service do not start... I tried to comment pieces of codes here and there to try to disable the parts about GPIOs in app.py like:
`from hardware.indicators import ApprovedIndicator, BlockedIndicator`
&
`        self.approved_indicator = ApprovedIndicator(
                self.config.get("GPIO_LED_APPROVED_PIN"),
                self.config.get("GPIO_LED_APPROVED_BRIGHTNESS", 100))
        self.blocked_indicator = BlockedIndicator(
                self.config.get("GPIO_LED_BLOCKED_PIN"),
                self.config.get("GPIO_LED_BLOCKED_BRIGHTNESS", 100))`

also in config.py all about GPIO_LED
But I still have the same error about gpiozero not accessible. I'm not sure I'm in the right direction...

I was wondering if it would be possible to first understand if changes that have to be made to get rid of GPIO is an hard job or if it can be made easily...

If you still visit this page let us know :)
Thanks a lot, merci beaucoup.
Best regards.

J0hnMatrix, 2022-09-10:
Hello advfr,
You should take a look at this commit for disabling GPIO:
https://github.com/emxsys/callattendant/pull/152

I don't had time to try it as I would like to install Callattendant on a Linux VM too.

Bye from France too ;-)

advfr, 2022-09-11:
Salut John, thanks a lot for your so prompt answer.
I took a look to the commit proposed by Lucien Van Elsen.
I'm really really far to be an expert, but it looks like that nowhere I see something to avoid the download of the... what... libraries (in setup.py) like gpiozero, pigpio, RPi.GPIO and RPIO.
In app.py, he still imports things from hardware.indicators. (indicators.py itself imports things from gpiozero)
I'm thinking changes have been suggested not to use GPIO but still using a Pi.
I don't know if I'm clear and if you see what I mean...
I created of fork and I'm trying to comment maximum things about indicators/GPIO things like that...
I left from galacticstudios's fork, which has an interesting improvement with possibility to manage white/black list with web interface and regular expressions.
All the best, I will come back here if I'm able to do what I want.
A bientôt !

advfr, 2022-09-12:
Hi again, I think my fork is functionnal.
https://github.com/advfr/callattendant/

Once again big thanks to Bruce !

J0hnMatrix, 2022-09-12:
Hello advfr,
Thank you very much for making Callattendant compatible with other OS than Raspberry OS.
I will take a look when I will have some time :)

Have a nice day!

foureight84, 2023-01-22:
I've attempted to dockerize @advfr's fork but I am currently having issues with calls not being detected. The modem initializes but I haven't yet debugged the reason for the detection issue. I'll update and if you know the reason, please feel free to share.

[https://github.com/foureight84/callattendant-docker](https://github.com/foureight84/callattendant-docker)

foureight84, 2023-01-24:
> I've attempted to dockerize @advfr's fork but I am currently having issues with calls not being detected. The modem initializes but I haven't yet debugged the reason for the detection issue. I'll update and if you know the reason, please feel free to share.
> 
> https://github.com/foureight84/callattendant-docker
> 
> EDIT: I can send commands and receive response from the modem via minicom inside the container. Debug log shows that app is detecting rings but no actions are taken.

I decided to fork the master branch and remove references to gpio usage. It's working now. You can find the updated branch here: https://github.com/foureight84/callattendant/tree/no-gpio

Docker instance working as well.

@advfr I believe your changes might have removed the ring event thread. This is just a hunch since callattendant never kicks in to handle the call during testing. I haven't reviewed the changes made.