emxsys/callattendant #139: Can't get any data through the USB modem

dimitriscr, 2020-12-03:
Hello, I have set up my callattendant like described in the setup guide but it doesn't seem to be able to see any phone calls made to the landline I have. I am in France and I'm using the "Zoom 56k Dial-up V.92 External USB Modem". Callattendant seems to detect the modem, but I can't tell if the following output means that it is working or not.

```
(venv) pi@raspberrypi:~ $ callattendant --config ~/.callattendant/app.cfg
Command line options:
  --config=/home/pi/.callattendant/app.cfg
  --data-path=None
  --create-folder=False
[Configuration]
  BLOCKED_ACTIONS = ('greeting', 'voice_mail')
  BLOCKED_GREETING_FILE = /home/pi/venv/lib/python3.7/site-packages/callattendant/resources/blocked_greeting.wav
  BLOCKED_RINGS_BEFORE_ANSWER = 0
  BLOCK_ENABLED = False
  BLOCK_NAME_PATTERNS = {'V[0-9]{15}': 'Telemarketer Caller ID'}
  BLOCK_NUMBER_PATTERNS = {}
  BLOCK_SERVICE =
  CONFIG_FILE = /home/pi/.callattendant/app.cfg
  DATABASE = callattendant.db
  DATA_PATH = /home/pi/.callattendant
  DB_FILE = /home/pi/.callattendant/callattendant.db
  DEBUG = True
  ENV = production
  GPIO_LED_APPROVED_BRIGHTNESS = 100
  GPIO_LED_APPROVED_PIN = 15
  GPIO_LED_BLOCKED_BRIGHTNESS = 100
  GPIO_LED_BLOCKED_PIN = 17
  GPIO_LED_MESSAGE_BRIGHTNESS = 100
  GPIO_LED_MESSAGE_COUNT_KWARGS = {'active_high': True}
  GPIO_LED_MESSAGE_COUNT_PINS = (8, 7, 27, 23, 10, 11, 9, 18)
  GPIO_LED_MESSAGE_PIN = 4
  GPIO_LED_RING_BRIGHTNESS = 100
  GPIO_LED_RING_PIN = 14
  PERMITTED_ACTIONS = ('greeting', 'record_message')
  PERMITTED_GREETING_FILE = /home/pi/venv/lib/python3.7/site-packages/callattendant/resources/general_greeting.wav
  PERMITTED_RINGS_BEFORE_ANSWER = 4
  PERMIT_NAME_PATTERNS = {}
  PERMIT_NUMBER_PATTERNS = {}
  PHONE_DISPLAY_FORMAT = ###-##-##-##
  PHONE_DISPLAY_SEPARATOR = -
  ROOT_PATH = /home/pi/venv/lib/python3.7/site-packages/callattendant
  SCREENED_ACTIONS = ('greeting', 'record_message')
  SCREENED_GREETING_FILE = /home/pi/venv/lib/python3.7/site-packages/callattendant/resources/general_greeting.wav
  SCREENED_RINGS_BEFORE_ANSWER = 0
  SCREENING_MODE = ('whitelist', 'blacklist')
  TESTING = False
  VERSION = 1.1.0
  VOICE_MAIL_GOODBYE_FILE = /home/pi/venv/lib/python3.7/site-packages/callattendant/resources/goodbye.wav
  VOICE_MAIL_GREETING_FILE = /home/pi/venv/lib/python3.7/site-packages/callattendant/resources/general_greeting.wav
  VOICE_MAIL_INVALID_RESPONSE_FILE = /home/pi/venv/lib/python3.7/site-packages/callattendant/resources/invalid_response.w                                                                                                                    av
  VOICE_MAIL_LEAVE_MESSAGE_FILE = /home/pi/venv/lib/python3.7/site-packages/callattendant/resources/please_leave_message.                                                                                                                    wav
  VOICE_MAIL_MENU_FILE = /home/pi/venv/lib/python3.7/site-packages/callattendant/resources/voice_mail_menu.wav
  VOICE_MAIL_MESSAGE_FOLDER = /home/pi/.callattendant/messages
Initializing Modem
Opening serial port
Looking for modem on /dev/ttyACM1
_send_and_read('AT','OK',5)
b'\r\n'
b'OK\r\n'
_send_and_read('ATI0','OK',5)
b'\r\n'
b'56000\r\n'
b'\r\n'
b'OK\r\n'
******* Conextant-based modem detected **********
Serial port opened on /dev/ttyACM1
Initializing modem settings
_send_and_read('ATZ','OK',5)
b'\r\n'
b'OK\r\n'
_send_and_read('ATV1','OK',5)
b'ATV1\r\r\n'
b'OK\r\n'
_send_and_read('ATE0','OK',5)
b'ATE0\r\r\n'
b'OK\r\n'
_send_and_read('AT+VCID=1','OK',5)
b'\r\n'
b'OK\r\n'
_send_and_read('AT&W0','OK',5)
b'\r\n'
b'OK\r\n'
_send_and_read('AT&V','OK',5)
b'\r\n'
b'ACTIVE PROFILE:\r\n'
(b'B1 E0 L2 M1 N0 Q0 T V1 W0 X4 Y0 &C1 &D2 &G0 &J0 &K3 &Q5 &R1 &S0 &T5 &X0 '
 b'&Y0\r\n')
(b'S00:000 S01:000 S02:043 S03:013 S04:010 S05:008 S06:002 S07:050 S08:001 S09:'
 b'006\r\n')
(b'S10:014 S11:085 S12:050 S18:000 S25:005 S26:001 S36:007 S38:020 S46:138 S48:'
 b'007\r\n')
b'S95:000 \r\n'
b'\r\n'
b'STORED PROFILE 0:\r\n'
b'B1 E0 L2 M1 N0 Q0 T V1 W0 X4 Y0 &C1 &D2 &G0 &J0 &K3 &Q5 &R1 &S0 &T5 &X0 \r\n'
(b'S00:000 S02:043 S06:002 S07:050 S08:001 S09:006 S10:014 S11:085 S12:050 S18:'
 b'000\r\n')
b'S36:007 S40:104 S41:195 S46:138 S95:000 \r\n'
b'\r\n'
b'STORED PROFILE 1:\r\n'
b'B1 E1 L1 M1 N0 Q0 T V1 W0 X4 Y0 &C1 &D2 &G0 &J0 &K3 &Q5 &R1 &S0 &T5 &X0 \r\n'
(b'S00:000 S02:043 S06:002 S07:050 S08:001 S09:006 S10:014 S11:085 S12:050 S18:'
 b'000\r\n')
b'S36:007 S40:104 S41:195 S46:138 S95:000 \r\n'
b'\r\n'
b'TELEPHONE NUMBERS:\r\n'
b'0=                                     1=\r\n'
b'2=                                     3=\r\n'
b'\r\n'
b'OK\r\n'
Modem initialized
Initializing CallLogger
CallLogger initialized
Initializing CallScreener
Initializing Blacklist
Blacklist initialized
Initializing Whitelist
Whitelist initialized
CallScreener initialized
Initializing VoiceMail
Initializing Message
Unplayed message count is 0
Message initialized
Resetting Message Indicator to show 0 unplayed messages
{MSG LED OFF}
VoiceMail initialized
Starting the Flask webapp
Running the Flask server
Waiting for call...
 * Serving Flask app "userinterface.webapp" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on http://0.0.0.0:5000/ (Press CTRL+C to quit)

```

I don't know what the problem could be, it might even just be the modem that doesn't work (I bought it used). Sadly I don't know enough about USB modems to interpret the tests that it seems to be doing. Oh and the web interface works without any problems.

Thx in advance for your help.

Strider2525, 2020-12-03:
Hi
I have found the same problem
I have installed the application (V1.1.0) as the guide and I am using the Zoom 56k USB Modem Series 1063 in the UK with caller ID enabled on the line
I get the same output as above when the app is run
I have found that the modem is not detecting the phone number. As my modem is used, I thought it might be the issue until I saw the post from dimitriscr
The output when I make an inbound call is as follows:

b'RING\r\n'
{RING LED BLINKING}
b'RING\r\n'
{RING LED BLINKING}
b'RING\r\n'
{RING LED BLINKING}
b'RING\r\n'
{RING LED BLINKING}
b'RING\r\n'
{RING LED BLINKING}

I have now purchased a US Robotics Model 5637 and when I make an inbound call I get the following output:

b'DATE=1203\r\n'
b'TIME=1452\r\n'
b'NMBR=07xxxxxxxx7\r\n'
> Queueing call 07xxxxxxxx7 for processing
Adding to caller queue:
{'DATE': '1203', 'NAME': 'Unknown', 'NMBR': '07xxxxxxxx7', 'TIME': '1452'}
Incoming call from 07xxxxxxxx7
> Checking whitelist(s)
>> Checking permitted patterns...
> Checking blacklist(s)
>> Checking blocked patterns...
Caller has been screened
> New call log entry #1

emxsys, 2020-12-03:
Hi @dimitriscr. I'm sorry you're having problems. Thanks for the DEBUG log; your modem initialization looks OK.  

Two possible issues come to mind:
1. Do you have caller ID service on your landline? (It's an optional pay for service in the USA).  
2. How is your modem connected to your phone system?

The fact that there are no "RING" entries in your DEBUG log implies that the modem is not actually receiving any signals from the phone system.  The modem should be connected jack that a phone could use. To test, you should be able to connect a phone in place of the modem to assert that the cable and jack are functional (e.g., assert a dial tone).  

I hope this helps. Let me know your progress.
--Bruce

dimitriscr, 2020-12-03:
My Modem is connected by a splitter to the same line as my landline that works properly. Caller_ID is slightly more of a wildcard. There is no mention of it anywhere on my providers plan or website. Since there is no way for me to pay for the service i just assumed i had it by default.  Since you said that the Modem seamed to work properly, i guess that leaves called_ID as the only thing left. I'll try to see if there is a way for me to confirm if I do have it or not.

TL:DR it's not the modem and my line probably doesn't have caller_ID enabled

Thx for your help

emxsys, 2020-12-03:
@Strider2525, thanks for your comment.  It would appear that the "Zoom 56k USB Modem **Series 1063**" does not appear to be a "voice modem".  I only found reference to "faxmodem" in the manual.  It be clear, your callattendant is now working correctly with the USR modem, correct?

The modems I've tested are in this thread:  https://groups.io/g/callattendant/topic/78477504#108  
Comments about modems that don't work would be useful.

@dimitriscr What is the model number of your Zoom modem?  I'm still suspicious about the lack of RING indications (like those in @Strider2525's comment). Also, I've used a splitter successfully before, so that by itself should not be an issue.  Do the modem lights/LEDs give any indication of an incoming call?

dimitriscr, 2020-12-03:
The exact Model is: "Zoom 56k DIAL-UP V.92 USB Modem", which i belive is the second one that in the list of supported models. 
[image]

The LEDs don't blink a lot, only a little bit when first powered, that was what led me to think that it might have been the modem not being properly detected in the first place.

emxsys, 2020-12-03:
@dimitriscr That's one of the modems that I have successfully tested with.  I will hook it up again today to verify what the correct behavior should look like.

Meanwhile, one last thing to try is to power cycle the modem: disconnect/reconnect the USB modem from the PI and then restart the call attendant.

Strider2525, 2020-12-03:
Hi Bruce

It is the correct Zoom modem, the series is 1063 and the model is 3095, picture attached
Sorry for the confusion
The US robotics works fine in the same setup and callattendant is working with it

I have checked on the Zoom, when an inbound call comes in only the DATA light flashes, the OH does not come on at all

[image]

emxsys, 2020-12-03:
@Strider2525, thanks for the clarification. Here's a picture of my Zoom 3095...not too much in common. 
[image]

dimitriscr, 2020-12-03:
@emxsys I have the square one too, could that be the issue?

Strider2525, 2020-12-03:
@emxsys it looks like yours is a different model, a 3095A, as well as its shape
So I can only think that is the issue, the 3095 does not work

emxsys, 2020-12-03:
@Strider2525 I concur.  

My modem box is identical to the picture posted by @dimitriscr.  Unfortunately, the "A" model doesn't appear to be a discriminator when ordering.

emxsys, 2020-12-03:
@dimitriscr Following is my debug log when using a 3095A modem.  The modem initialization looks identical to yours, as does the active profile's modem registers.

Do you have a "distinctive ring" enabled on your phone?  See **_+VDR - Distinctive Ring_** in this document: http://www.zoomtel.com/documentation/dial_up/3095F_ATcommands.pdf.  Could that be a factor?

If your familiar with minicom or another terminal program, then you can experiment with the modem AT commands needed to get the RING notification to work.  

#### My modem init with 3095A
```
Initializing Modem
Opening serial port
Looking for modem on /dev/ttyACM0
_send_and_read('AT','OK',5)
b'\r\n'
b'OK\r\n'
_send_and_read('ATI0','OK',5)
b'\r\n'
b'56000\r\n'
b'\r\n'
b'OK\r\n'
******* Conextant-based modem detected **********
Serial port opened on /dev/ttyACM0
Initializing modem settings
_send_and_read('ATZ','OK',5)
b'\r\n'
b'OK\r\n'
_send_and_read('ATV1','OK',5)
b'\r\n'
b'OK\r\n'
_send_and_read('ATE0','OK',5)
b'\r\n'
b'OK\r\n'
_send_and_read('AT+VCID=1','OK',5)
b'\r\n'
b'OK\r\n'
_send_and_read('AT&W0','OK',5)
b'\r\n'
b'OK\r\n'
_send_and_read('AT&V','OK',5)
b'\r\n'
b'ACTIVE PROFILE:\r\n'
(b'B1 E0 L2 M1 N0 Q0 T V1 W0 X4 Y0 &C1 &D2 &G0 &J0 &K3 &Q5 &R1 &S0 &T5 &X0 '
 b'&Y0\r\n')
b'S00:0 S01:0 S02:43 S03:13 S04:10 S05:8 S06:2 S07:50 S08:1 S09:6\r\n'
b'S10:14 S11:85 S12:50 S18:0 S25:5 S26:1 S36:7 S38:20 S46:138 S48:7\r\n'
b'S95:0 \r\n'
b'\r\n'
b'STORED PROFILE 0:\r\n'
b'B1 E0 L2 M1 N0 Q0 T V1 W0 X4 Y0 &C1 &D2 &G0 &J0 &K3 &Q5 &R1 &S0 &T5 &X0 \r\n'
b'S00:0 S02:43 S06:2 S07:50 S08:1 S09:6 S10:14 S11:85 S12:50 S18:0\r\n'
b'S36:7 S40:104 S41:203 S46:138 S95:0 \r\n'
b'\r\n'
b'STORED PROFILE 1:\r\n'
b'B1 E1 L2 M1 N0 Q0 T V1 W0 X4 Y0 &C1 &D2 &G0 &J0 &K3 &Q5 &R1 &S0 &T5 &X0 \r\n'
b'S00:0 S02:43 S06:2 S07:50 S08:1 S09:6 S10:14 S11:85 S12:50 S18:0\r\n'
b'S36:7 S40:104 S41:203 S46:138 S95:0 \r\n'
b'\r\n'
b'TELEPHONE NUMBERS:\r\n'
b'0=                                     1=\r\n'
b'2=                                     3=\r\n'
b'\r\n'
b'OK\r\n'
Modem initialized
```

This is the debug log I get when a call comes in:
```
b'RING\r\n'
{RING LED BLINKING}
b'DATE = 1203\r\n'
b'TIME = 1042\r\n'
b'NMBR = xxxxxxxxxx\r\n'
b'NAME = BRUCE SCHUBERT\r\n'
> Queueing call xxxxxxxxxx for processing
Adding to caller queue:
{'DATE': '1203', 'NAME': 'BRUCE SCHUBERT', 'NMBR': 'xxxxxxxxxx', 'TIME': '1042'}
Incoming call from xxxxxxxxxx
> Checking whitelist(s)
>> Checking permitted patterns...
> Checking blacklist(s)
>> Checking blocked patterns...
>> Checking nomorobo...
Caller has been screened
> New call log entry #2735
['BRUCE SCHUBERT',
 'xxxxxxxxxx',
 'Screened',
 'Not found',
 '03-Dec',
 '10:42 AM',
 '2020-12-03 10:42:12']
--> xxxxxxxxxx Screened: Not found
> Going off hook...

```

emxsys, 2020-12-03:
@dimitriscr @Strider2525, per table 3.5 in http://www.zoomtel.com/documentation/dial_up/3095F_ATcommands.pdf:

> RING An incoming ring signal is detected on the line. What qualifies as a ring signal is determined by
> country-dependent parameters. 

Here's something to look into with a terminal program: 

### +GCI - Country of Installation
This extended syntax command selects and indicates the country of installation for the
modem. This parameter selects the settings for any operational parameters that need to be
adjusted for national regulations or telephone networks.

#### Syntax
`+GCI=<country_code>`

#### Reporting Current or Selected Values
- Command: `+GCI?`
- Response: `+GCI: <current country_code>`
- Example: `+GCI: 3D` The modem is set for France.

dimitriscr, 2020-12-03:
Alright, sorry for the wait, it took me a bit to understand how minicom worked. My GDR is B5 which i assume is not France and my VDR is 0,0 which I assume means it false. Is there a way to change the GDR to france or is it build into the router? Is it what is preventing my modem from receiving calls?

emxsys, 2020-12-03:
Try: +GCI=3D

B5 (USA) is the factory default.

Bruce Schubert
KM6ZJK
805.207.1080
bruce@emxsys.com

dimitriscr, 2020-12-03:
Alright, I changed it, I checked it again and it has indeed changed to 3D, but still no luck with getting it to detect a ring.

emxsys, 2020-12-03:
Rats! I was hopeful that was it. 
Do you think distinctive ring could be a factor?

dimitriscr, 2020-12-03:
[image]
Mine is set to 0,0. There are a few mentions of "will not report ring...", but I am not sure as to how revelevant to my issue. should I change it to 1,1?

emxsys, 2020-12-03:
It's definitely worth trying.  It's a data point either way.

dimitriscr, 2020-12-03:
Hell yeah, that was it!
[image]
Thanks a lot for your help, I ended up learning a bit about USB modem as a side effect too!

dimitriscr, 2020-12-03:
[image]
I have this parameter on 1, this is what it's meant to be right?

emxsys, 2020-12-03:
Great news! I'll consider how to factor this into the codebase or configuration.

Re: CID. You will need to enable it. I think this is the command:  AT+VCID=1

emxsys, 2020-12-03:
To be clear you would need to enable VCID when testing in minicom. The callattendant enables it in the modem initialization...you can see it in your debug log.

dimitriscr, 2020-12-03:
Alright, I still don't seam to be able to get a number, I will keep reading that documentation you send me and i'll see if I can fix it tomorrow.

Thx again for your help

Tikkoneus, 2023-01-16:
@emxsys , @dimitriscr I'm having a similar issue of never seeing a RING, but none of the above ideas have helped. I tried explicitly setting GCI=B5 and VDR=1,1 and the modem init looks fine but then no ring ever shows up. Any other ideas?