emxsys/callattendant #142: Missing lxml parser library

anonyme22, 2020-12-26:
Dec 26 11:58:26 callattendant[8290]: {RING LED BLINKING}
Dec 26 11:58:27 callattendant[8290]: > Queueing call 14387883368 for processing
Dec 26 11:58:27 callattendant[8290]: Incoming call from 14387883368
Dec 26 11:58:27 callattendant[8290]: > Checking whitelist(s)
Dec 26 11:58:27 callattendant[8290]: >> Checking permitted patterns...
Dec 26 11:58:27 callattendant[8290]: > Checking blacklist(s)
Dec 26 11:58:27 callattendant[8290]: >> Checking blocked patterns...
Dec 26 11:58:27 callattendant[8290]: >> Checking nomorobo...
Dec 26 11:58:27 callattendant[8290]: **FeatureNotFound("Couldn't find a tree builder with the feature                             s you requested: lxml. Do you need to install a parser library?")**
Dec 26 11:58:27 callattendant[8290]: ** Error running callattendant
Dec 26 11:58:27 callattendant[8290]: Shutting down...
Dec 26 11:58:27 callattendant[8290]: -> Stopping modem
Dec 26 11:58:28 callattendant[8290]: -> Closing modem serial port
Dec 26 11:58:28 callattendant[8290]: -> Stopping voice mail
Dec 26 11:58:29 callattendant[8290]: -> Releasing resources
Dec 26 11:58:29 callattendant[8290]: Shutdown finished
Dec 26 11:58:29 systemd[1]: callattendant.service: Main process exited, code=exited, status=1/FAILU                             RE
Dec 26 11:58:29 systemd[1]: callattendant.service: Failed with result 'exit-code'.

but it's already install : 
pi@raspberrypi:~/.callattendant $ pip3 install lxml
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: lxml in /home/pi/.local/lib/python3.7/site-packages (4.6.2)

owencrow, 2021-01-29:
I was able to fix on my "Raspbian GNU/Linux 10 (buster)" with 

`apt install python-bs4`

nomorobo.py is still Python2 so the lxml library is in the package above. This is probably solvable via a pip command, too, but got the one I used from the commit comments that added the lxml line to the code: [https://github.com/emxsys/callattendant/commit/e990aa117346af15dbb8bd159f799bf948ac526d](https://github.com/emxsys/callattendant/commit/e990aa117346af15dbb8bd159f799bf948ac526d)

anonyme22, 2021-03-21:
Thank owencrow, 

This is working.

closed 2021-03-21 by anonyme22