emxsys/callattendant #147: Modem get's recognized but does not give any number

pabloemma, 2021-03-13:
Just installed the system on a Raspberry pi3B with a us robotics 5637 and callattendant recognizes the modem.
When I call, the modem will even blink on the data, and the output in debug just shows
b'RING\r\n'
{RING LED BLINKING}
b'RING\r\n'
{RING LED BLINKING}
I have caller ID on my line and the jack I am using works. I tested it with a real phone.
I attach my output.
I am running python3.7.3 
Thanks for any help,
Andi

[debug.txt](https://github.com/emxsys/callattendant/files/6133732/debug.txt)

marklt1, 2022-03-27:
> Just installed the system on a Raspberry pi3B with a us robotics 5637 and callattendant recognizes the modem. When I call, the modem will even blink on the data, and the output in debug just shows [...]

I am seeing the exact same behavior also. My debug text was like yours.

pabloemma, 2022-03-27:
I never got the us robotics to work,so I got the zoom one. zoom 56k usb
modem
that works great with my raspberry pi
andi

marklt1, 2022-03-27:
> 

Thanks for the tip. I was thinking now that it could be a python version issue but now likely a hardware issue based on your experience.