emxsys/callattendant #159: virtual build failing

mikejrh, 2021-12-18:
I get the following error message when trying to install callattendant in the virtual environment

Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/venv/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-svnjf43h/rpi-gpio_50fc5284668e4483af35739ed49ea1a5/setup.py'"'"'; __file__='"'"'/tmp/pip-install-svnjf43h/rpi-gpio_50fc5284668e4483af35739ed49ea1a5/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-489lebh_
       cwd: /tmp/pip-install-svnjf43h/rpi-gpio_50fc5284668e4483af35739ed49ea1a5/

Peter-Gitman, 2022-01-18:
I also got the same error message.  Not sure where to turn from here.

tagatac, 2022-01-25:
It seems like RPi.GPIO does not follow modern linking conventions, so gcc complains. I was able to install by telling gcc that this is legacy code like this:
```
CFLAGS=-fcommon pip install RPi.GPIO
```

References:
1. https://askubuntu.com/a/1330210/267107
2. https://gcc.gnu.org/onlinedocs/gcc/Code-Gen-Options.html