emxsys/callattendant #161: NO Issue...just had to say thanks.

jumper-d-1981, 2022-02-02:
You can absolutely close this after you get it.  

I just wanted to say thank you.  Is there a way to donate to this project?  I am in love.  You have it working essentially out of the box.  I just plugged everything together and ran the first command or two and it started working right then and there.  I absolutely love it.  I will be tightening it up a bit...for security, but this is truly amazing.  It has been 9 years of me searching for a way to block these freaking people...I don't even care it has to ring once before action is taken...but I will say this::  Thanks, and much love!

zomGreg, 2022-03-06:
Hah, glad to hear you like the project. I also am a huge fan. Please contribute in any way you see fit.

bdmillan, 2023-01-23:
https://www.vccomm.org/km6zjk-bruce-schubert-sk/
Hello, I thought I should put this out there for callattendant fans. It is also on the forum. 
https://groups.io/g/callattendant/message/344