emxsys/callattendant #168: First Ring Suppressor...UPDATE!

jumper-d-1981, 2022-09-16:
Hello, 
I noticed this Call Attendant works well, but first ring still happens.  This annoys me when I am woken up at 4am to a tele marketer single ring...lol.  Sometimes two.

I know some folks bought NOMOROBO, just to get the first ring to shut up.  BUT I did a ton of digging and found this from a man who worked for a company back in the 90s, and he designed this.  Since they no longer deal with this technology in his part of the world(all digital and voip) he just gave this to me hoping I could run with it.  BUT, I have not attempted it yet.  Thought one of the people in here might be able to put it to use?

It is a BOM and schematic for a first ring supressor.  I sent this to PCB way, they said it would cost less than 4 bucks a piece in small batches if anyone wanted to make them.  If you do credit Brad Peeters would be great...especially if you sell them.

I think it may need updated because voltages I do not believe are the same today on POTS as they were in the 90s.

Anyway.  enjoy.
[FRS_BOM.xlsx](https://github.com/emxsys/callattendant/files/9588395/FRS_BOM.xlsx)
[FRS Schematic.PDF](https://github.com/emxsys/callattendant/files/9588397/FRS.Schematic.PDF)

anonyme22, 2022-11-27:
I fond this :
https://www.airplgri.com/frs22100-first-ring-suppressor-silencer-for-robocall-blocking-services-p-449385.html?products_id=449385:c05cf5a1c0745d85af67593931ac8314
3.25$ USD. Look nice... but in my case, I'm from Canada and it will cost me 20$ USD shipping...

jumper-d-1981, 2022-11-28:
> Hi! Did you have a chance to try it? I belive the voltages will be still the same as was in the 90s.

No, I cannot source one of the components.  Looks like it was deprecated many many years ago.  That is, I can find the part, but it is expensive...40 bucks.  So I did not look any further.  HOWEVER, I did need a new phone so I just got one with a option to block the first ring. Which works perfectly.  This in conjunction with Call Attendant...I have had zero actual bull crap phone calls and only once in a while do I hear a single ring.  AND since the caller id still fires even though it stops the ring audibly I can see if there is a number accidentally getting swept aside by the system.  I do still get phone calls from people in my area, but they are not actually the ones calling.  AND I still get those stupid polls.  BUT after adjusting the rules and black list thing I seem to catch 90-93% of the bs.

It has been a god send.

As to the device you found.  That is what I was looking for originally.  I could not find it in stock.  If you can get it, the price is great.

Let us know how it goes.  Not everyone can drop 100 bucks on a new 3 phone system...and I would not have except my other phone's "main base" took a dump on me.  I had it working in my home for 15 years so it was probably time.

Good luck.

GrahamM, 2023-02-22:
Just a warning for anyone that tries to build this. The first ring suppressor listed relies on a microcontroller (PIC12C508) to operate. It'd be easy to substitute an in-production part for that. However, the microcontroller needs firmware to operate. Without the firmware, the schematic isn't going to do a lot sadly.

almottier, 2023-04-27:
Hello. Has anyone removed the first ring with a software or hardware solution?
In the README, it says `that intercepts robocallers and blocked numbers at or before the first ring`, how is this possible?

jumper-d-1981, 2023-04-27:
As I understand it.  The data that contains number and name(id caller) is transmitted before the first ring...but based on the ticks of the switching mechanics(whatever hardware makes phones work) sometimes that information is sent before the first ring actually makes a sound and sometimes during the ringing.  If things aren't working right on my phone line sometimes it doesn't send at all or until a 2nd or 3rd ring.

BUT typically the packet of data or whatever that is sent is sent during that first or before the first ring.  The software in this attendant software intercepts that data just like an id caller and if it matches white or black listed numbers it handles it accordingly.

Does that make sense?

I am uncertain this software will work on voip, but it does work on pots and digital phone...which is different than voip.

Also has anyone blocked the first ring?  YES.  Some folks buy the el-cheapo robo blockers from amazon that will suppress the ring.  ID info is still sent normally but the noise is silenced.  Some modern phones have the ability to silence the first ring as well.  

I found a piece of hardware(design) from the 90s from a man who gave it to us to use to create our own suppressors but when I talked to PCB way about it, we could not source 1 of the components, and that component would need to be programmed with a IC programmer that is beyond my skills.

BUT, it is possible to silence all your phone rings, then add a software ringer that rings on a speaker attached to the GPIO of the pi that runs this software.  Perhaps a possible solution for those with only one phone  or no need to hear it in multiple places.

It would be fairly easy to also attach other Micro Controllers like an ESP32 to ring remotely.  BUT it is a longer process than I am willing to do.

I use this software, it works great.  I did not add in anything other than a few custom messages, and a couple number rules.  And a light to flash when ringing.  Other than that, it is stock...and just works.

alright, this was longer than I expected.  I had coffee.  :)

Regards.