emxsys/callattendant #172: A few requests...and observations.

jumper-d-1981, 2023-02-20:
I think I read that the author passed away?!  If this is correct, my condolences to his friends and family.  

This is a really neat piece of software, and a great little community.  Thanks for allowing me to opportunity to be part of it.

I was wondering if anyone knew the answer to a couple things.
-How do I find the version of the attendant I am currently running?
-Has anyone considered installing auto update or update alerts via the web GUI such as klipper, octoprint, and pi-hole does?  Just a simple alert saying "new stuff" click here to find out how to install.
-has anyone figured out a way to upload custom messages?  I would like to use my own greeting when a call is blocked...the generic one is okay, but I wanted to be a little more spicy.  

Any insight is appreciated.

Thanks.

roycewilliams, 2023-03-14:
@jumper-d-1981 For the upload of custom messages ... there's no _GUI_ way to do it today, but you can `scp` replacement files and drop them into the `.callattendant/notifications` directory. They need to be 8-bit / 8-KHz mono unsigned PCM WAV files, which you can produce with Audacity (see step 3 [here](https://iotbytes.wordpress.com/play-audio-file-on-phone-line-with-raspberry-pi/)

thess, 2023-06-27:
@jumper-d-1981 - I'd be happy to consider any/all suggestions to improve this wonderful application. Check out my [work ](https://github.com/thess/callattendant)as @roycewilliams mentioned.

jumper-d-1981, 2023-06-27:
@thess ...
I do not have a ton of time these days, but some obvious things come to mind.
This software works great, but if we cannot get the OG account, then we should create one or sticky an announcement(if possible) so we can move over to a fork that would be maintained by you or us.

Some wish list items:
Would be nice to have a place to easily modify our lists.  Manually adding one at a time on the gui is fine, but sharing lists, or compiling a regional comprehensive list would be just darn nice...and the ability to turn them on or add them easily would be sweet.

Would be nice to also have the ability to easily modify the greetings and announcement recordings...manually takes a bit of effort and I fought and fought with the quality being terrible.

Would also be nice to get a hat for the pi made.  I can handle that if I know it is working correctly in prototype stage.  I could place the design up on PCBway or JCLPCB so others can buy a pre-built one for a few bucks.  I had a prototype kinda working...Red light on block, green light on allow, blue light when ringing...and stopped there, but wanted to add a "mute" switch to disable the light with a button.  I also wanted the RGB light to cycle through the colors when there was a message waiting every few seconds.

Dark theme for the gui would be sweet too.

BUT that is a wish list. 

Thank you for any and all that you do.  I know I appreciate any effort, as do others in this group.

thess, 2023-06-28:
@jumper-d-1981 - I don't think GH will just hand-over an account or even a repo without the author or their representative requesting/approving it. The best we can do in this situation is to fork the repo and post an issue (large caps) noting that this repo is deprecated. Alternatively, you might try contacting Bruce's outfit (emxsys.com) and see if they have access to the repo. Another issue will be the PyPi account the callattendant package is currently distributed from.

> Would be nice to have a place to easily modify our lists. Manually adding one at a time on the gui is fine, but sharing lists, or compiling a regional comprehensive list would be just darn nice...and the ability to turn them on or add them easily would be sweet.

My version has the ability to save and restore lists in CSV format.

> Would be nice to also have the ability to easily modify the greetings and announcement recordings...manually takes a bit of effort and I fought and fought with the quality being terrible.

You can store any custom **.wav** file in the `notifications` folder as @roycewilliams mentioned above. The filenames can be referenced in the config file. There are several good (free) text-to-speech services available. You will need to convert the output to the format required by the modem you are using.

> Dark theme for the gui would be sweet too.

PRs welcome :)